UT-Austin-RPL/GIGA #1: generate_data_parallel.py hangs with pool.apply_async(), but not when called in the main thread.

djl11, 2021-04-10:
By naively following the installation instructions, I'm finding that [this line](https://github.com/UT-Austin-RPL/GIGA/blob/main/src/vgn/perception.py#L105) hangs when running with multiprocessing, but not when running in the main thread.

Would it be possible to list the full conda environment you used, by running `conda list -e > requirements.txt`?
For example, I would be interested to know which versions of python, Open3D and multiprocessing you used.

Any help appreciated :)

Steve-Tod, 2021-04-11:
Hi, I add the conda environment description [here](https://github.com/UT-Austin-RPL/GIGA/blob/main/environment.yaml). BTW, I suggest debugging the data generation process with `--num-proc 1` argument to set it to single process.

djl11, 2021-04-12:
Thanks for sending over! Unfortunately using your conda environment didn't help, and setting `--num-proc 1` still causes it to hang. The only way I can get it to run is remove `mp.Pool()` entirely and call `main()` explicitly. I'll leave this issue open and further investigate, and then close with the solution once I work out what's going on.

Steve-Tod, 2021-04-12:
Sorry about that. What script are you running? `generate_data_parallel.py` or `construct_dataset_parallel.py`?

djl11, 2021-04-12:
`generate_data_parallel.py`. Could still just be something silly on my end, will post my findings either way!

djl11, 2021-04-12:
Multiprocessing issues with Open3D seem relatively common: [a](https://stackoverflow.com/questions/61618467/trouble-working-with-open3d-pythons-multiprocessing-and-feature-detection), [b](https://github.com/intel-isl/Open3D/issues/389), [c](https://github.com/intel-isl/Open3D/issues/1552).

I took a look at how Open3D apply parallelism for batch-processing in their own [examples](http://www.open3d.org/docs/release/tutorial/reconstruction_system/make_fragments.html?highlight=multiprocessing#batch-processing), and they use [joblib](https://github.com/joblib/joblib).

Replacing [lines](https://github.com/UT-Austin-RPL/GIGA/blob/main/scripts/generate_data_parallel.py#L195) 195-202 with:

    from joblib import Parallel, delayed
    Parallel(n_jobs=parsed_args.num_proc)(delayed(main)(parsed_args, i) for i in range(parsed_args.num_proc))`

Fixed the problem for me. Hopefully this helps others who may get stuck with similar issues.

closed 2021-04-12 by djl11

Andyyoung0507, 2024-09-12:
Hi, I also try `python scripts/generate_data_parallel.py --scene pile --object-set pile/train --num-grasps 4000000 --num-proc 40 --save-scene ./data/pile/data_pile_train_random_raw_4M` to generate the data, but the terminal show nothing but `pybullet build time:……`, I also set --num-proc to 1 to debug, but it didn't work. Then I use vscode to debug, find that is maybe caused by multiprocess. I tried @djl11 's solution, but nothing changed. So i reinstall the conda env, The script runs, but gets stuck somewhere after generating some data. May I have your help please?
Thanks for help. @djl11 @Steve-Tod