UT-Austin-RPL/GIGA #3: Downloadable Training Data

djl11, 2021-04-12:
It would be very useful to have the training data returned from `generate_data_parallel.py` script available to download, for both the pile and packed cases.

I appreciate this may be a large amount of memory, and therefore difficult to host, so there is no expectation of course!
But it would avoid people needing to run the costly data generation process locally in order to experiment with the training.

Steve-Tod, 2021-04-13:
Hi, I upload the link of data we generated [here](https://github.com/UT-Austin-RPL/GIGA#pre-trained-models-and-pre-generated-data). Hopefully it would be helpful.

djl11, 2021-04-13:
Great, thanks a lot!

closed 2021-04-13 by djl11

djl11, 2021-04-22:
Hey, quick heads up. The links in the table for the README are mis-matched. The pile links leads to packed data, and vice versa. Also a small type for the word "this" beforehand. Both should be easy to fix! :)

Steve-Tod, 2021-04-23:
Thank you for reminding me!

Andyyoung0507, 2024-09-06:
When using a browser to download a dataset, the download process is often interrupted due to network problems. I tried to download it using aria2, but the download link was not recognized successfully. Are there any other ways to download the dataset?
Thanks for your excellent work!

Steve-Tod, 2024-09-06:
Hi @Andyyoung0507 I added direct download link in [https://github.com/UT-Austin-RPL/GIGA/blob/main/README.md#pre-generated-data](https://github.com/UT-Austin-RPL/GIGA/blob/main/README.md#pre-generated-data). You can try download the data with `wget`.

Andyyoung0507, 2024-09-07:
Thanks for all your help!