UT-Austin-RPL/GIGA #4: Segmentation fault (core dumped)

ccc1711, 2021-06-25:
Hi,thank you for you repo!
However, I encountered an error when running  sim_grasp_multiple.py.
As shown in the figure 
[image]

Could you give me some suggestions？

Steve-Tod, 2021-06-25:
Sorry, I can't see why this is happening without any traceback. Maybe try `pdb` and see where this error comes from?

ccc1711, 2021-06-25:
I try to use ipdb to find out where the problem is，I found that when running to line 86 or line 90, the error will occur 
[image]

ccc1711, 2021-06-25:
More specifically
[image]

ccc1711, 2021-06-25:
It's very interesting.I solved this problem by installing open3d==0.12.0, the previous version was 0.13.0.

Steve-Tod, 2021-06-25:
Great! I was using open3d==0.12.0. (https://github.com/UT-Austin-RPL/GIGA/blob/a60495be5091e08019da66194adaf9588abb4352/environment.yaml#L101) Let me add this to README.

closed 2021-06-25 by Steve-Tod

wuzeww, 2021-07-07:
> It's very interesting.I solved this problem by installing open3d==0.12.0, the previous version was 0.13.0.

hello! I want to know that if your experimental results are consistent with the success rate in the paper. The results i reproduced were 5% lower than the paper.

ccc1711, 2021-07-07:
> > It's very interesting.I solved this problem by installing open3d==0.12.0, the previous version was 0.13.0.
> 
> hello! I want to know that if your experimental results are consistent with the success rate in the paper. The results i reproduced were 5% lower than the paper.
Hello！I also encountered this problem you mentioned. But it seems that my success rate is lower in the sence of pile and packed.Only when the object is blocks, the success rate will be the same as in the paper

Steve-Tod, 2021-07-08:
Hi guys, are you using the pre-trained model I shared or you train the model again by yourself?

wuzeww, 2021-07-08:
hi, I verified it with the model trained by myself. The result is 5% lower than the paper. And i also used the model you shared, the results are almost close the paper.

wuzeww, 2021-07-08:
> Hi guys, are you using the pre-trained model I shared or you train the model again by yourself?

I found some interesting things. When I only use the CPU to train 5 epochs, the result is better than 10 epochs under the GPU. The training parameters are the same as those given in the project

Steve-Tod, 2021-07-08:
> > Hi guys, are you using the pre-trained model I shared or you train the model again by yourself?
> 
> I found some interesting things. When I only use the CPU to train 5 epochs, the result is better than 10 epochs under the GPU. The training parameters are the same as those given in the project

I've never use CPU for training. And the model I released was trained for 20 epoches. I just found the default argument is 1- epoches. That might be the problem😂.

wuzeww, 2021-07-08:
> > > Hi guys, are you using the pre-trained model I shared or you train the model again by yourself?
> > 
> > 
> > I found some interesting things. When I only use the CPU to train 5 epochs, the result is better than 10 epochs under the GPU. The training parameters are the same as those given in the project
> 
> I've never use CPU for training. And the model I released was trained for 20 epochs. I just found the default argument is 10 epoches. That might be the problem.

okay, thank you very much.