UT-Austin-RPL/GIGA #6: ValueError: file_type not specified!

ccc1711, 2021-08-12:
Hi,
When I run sim_grasp_multiple.py and add the parameter --vis for simulation grasping, the following problem occurred:

(giga) wang@wang-U:~/github/GIGA$  python scripts/sim_grasp_multiple.py --num-view 1 --object-set pile/test --scene pile --num-rounds 100 --sideview --add-noise dex --force --best --model /home/wang/github/GIGA/data/models/giga_pile.pt --type giga --result-path /home/wang/github/GIGA/data/result --vis

pybullet build time: May 28 2020 16:37:34
Loading [giga] model from /home/wang/github/GIGA/data/models/giga_pile.pt
  0%|                                                                                                                                                     | 0/100 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "scripts/sim_grasp_multiple.py", line 123, in <module>
    main(args)
  File "scripts/sim_grasp_multiple.py", line 38, in main
    success_rate, declutter_rate = clutter_removal.run(
  File "/home/wang/github/GIGA/src/vgn/experiments/clutter_removal.py", line 89, in run
    logger.log_mesh(scene_mesh, visual_mesh, f'round_{round_id:03d}_trial_{trial_id:03d}')
  File "/home/wang/github/GIGA/src/vgn/experiments/clutter_removal.py", line 177, in log_mesh
    aff_mesh.export(self.mesh_dir / (name + "_aff.obj"))
  File "/home/wang/anaconda3/envs/giga/lib/python3.8/site-packages/trimesh/scene/scene.py", line 842, in export
    return export.export_scene(
  File "/home/wang/anaconda3/envs/giga/lib/python3.8/site-packages/trimesh/exchange/export.py", line 210, in export_scene
    raise ValueError('file_type not specified!')
ValueError: file_type not specified!

What is the reason?
Thank you！ ^_^

ccc1711, 2021-08-13:
I update trimesh to the latest version and it still doesn’t work, and I print out the path of **self.mesh_dir / (name + "_aff.obj")**,is **PosixPath(‘data/experiments/21-08-13-20-33-46/meshes/round_000_trial_000_aff.obj')** .How to specify the **file_type='obj'** ?

Steve-Tod, 2021-08-13:
You can try `aff_mesh.export(self.mesh_dir / (name + "_aff.obj"), file_type='obj')`. If this still does not work, actually the visualization is not necessary for running, and you can remove the argument `--vis`.

wuzeww, 2021-09-04:
> I update trimesh to the latest version and it still doesn’t work, and I print out the path of **self.mesh_dir / (name + "_aff.obj")**,is **PosixPath(‘data/experiments/21-08-13-20-33-46/meshes/round_000_trial_000_aff.obj')** .How to specify the **file_type='obj'** ?

Hi，
excuse me, have you realized the function of visualization?

ccc1711, 2021-09-04:
> Hi，
> excuse me, have you realized the function of visualization?

Not yet, how about you?

wuzeww, 2021-09-06:
> You can try `aff_mesh.export(self.mesh_dir / (name + "_aff.obj"), file_type='obj')`. If this still does not work, actually the visualization is not necessary for running, and you can remove the argument `--vis`.

Excuse me, I want to achieve the visualization results (Visualization of the grasp affordance landscape and predicted grasps) in your paper.  Does the code you upload have this function?

wuzeww, 2021-09-06:
> Not yet, how about you?

not yet

Steve-Tod, 2021-09-06:
> Excuse me, I want to achieve the visualization results (Visualization of the grasp affordance landscape and predicted grasps) in your paper. Does the code you upload have this function?

Yes, the code will save the visualization to `obj` file if you use the argument `--vis`. However, I don't know why you meet error when running `aff_mesh.export`.

wuzeww, 2021-09-09:
I tried **aff_mesh.export(str(self.mesh_dir / (name + "_aff.obj")))**, the issue of file_type disappeared.

Then i appeared array out of bounds (IndexError: index 201 is out of bounds for axis 0 with size 201).

Finally, found that the data of **data/urdfs/packed/test/Paprika_800_tex_visual.obj** is short of the letter 'v'(at the line of 15).

Steve-Tod, 2021-09-12:
> I tried **aff_mesh.export(str(self.mesh_dir / (name + "_aff.obj")))**, the issue of file_type disappeared.
> 
> Then i appeared the issue of **array out of bounds** (IndexError: index 201 is out of bounds for axis 0 with size 201).
> 
> Finally, found that the data of **data/urdfs/packed/test/Paprika_800_tex_visual.obj** is short of the letter 'v'(at the line of 15).

Hmm, that's interesting. Because I checked the data that I uploaded, and there's no such problem.
[image]

wuzeww, 2021-09-13:
> Hmm, that's interesting. Because I checked the data that I uploaded, and there's no such problem.
> [image]

I just re-extracted the data.zip file and found that the data is complete.

I want to know how long have you cost about training vgn one epoch. I performed on a computer with a GTX 1080 Ti graphics card and  the memory is 32G, but my cpu is noot good. I cost about 14 hours one epoch. I want to know if it’s the cpu problem.

Steve-Tod, 2021-09-13:
> I just re-extracted the data.zip file and found that the data is complete.
>
> I want to know how long have you cost about training vgn one epoch. I performed on a computer with a GTX 1080 Ti graphics [...]

When training the network on pile dataset (the larger one), one epoch takes 2 hours on one RTX 2080 Ti. So I think the problem is with CPU. You can check the "Volatile GPU-Util" given by `nvidia-smi` to see if the GPU is fully used. GPU-Util over 80% is a good sign.

huaji98, 2021-11-03:
> I just re-extracted the data.zip file and found that the data is complete.
>
> I want to know how long have you cost about training vgn one epoch. I performed on a computer with a GTX 1080 Ti graphics [...]

Hi,I also encountered the problem of slow training, I suspect that the slow IO speed of the mechanical hard drive is the reason, have you solved this problem?