UT-Austin-RPL/GIGA #9: Training speed is too slow

huaji98, 2021-11-03:
Hi,I training one epoch on pile scene,and estimated time to spend more than 10 hours.

Here is my computer configuration：
pytorch 1.7.0    py3.8_cuda10.2.89_cudnn7.6.5_0
GPU:GeForce RTX 2070SUPER
CPU:Intel(R) Core(TM) i7-9700 CPU @ 3.00GHz
Memory:16G
Hard Disk Drive:1T HDD

Here are some screenshots:
[image]
[image]
[image]
[image]

I found that GPU utilization was erratic and was 0 for a long time, and CPU utilization was not very high. So I presume it's because the io speed of the HDD hard disk drive is too slow.

I hope you can help me find the problem, thanks!

Steve-Tod, 2021-11-03:
I guess you are right, the speed of io is too slow. One thing you can try to verify this guess is to iterate the dataloader without training and check the speed.

huaji98, 2021-11-04:
yes,I found that loading the data took about the same amount of time as before with training.

closed 2021-11-04 by huaji98

Steve-Tod, 2021-11-05:
Do you get a similar speed when training giga-aff? If not, then the IO bottleneck is the occupancy data loading. The default size of each occupancy data file is 100K points, but we only sample 2048 points. So one thing you can try is to decrease the size and increase the number of occupancy files when generating data:

`python scripts/save_occ_data_parallel.py /path/to/raw/data {number_of_points_per_file} {number_of_files} --num-proc 40`

huaji98, 2021-11-08:
> Do you get a similar speed when training giga-aff? If not, then the IO bottleneck is the occupancy data loading. The default size of each occupancy data file is 100K points, but we only sample 2048 [...]

Thank you for your reply. I found a more powerful computer, and now the training speed is normal！