UT-Austin-RPL/GIGA #10: Training issue

liuallen871219, 2021-12-11:
I follow the command `python scripts/train_giga.py --dataset /path/to/new/data --dataset_raw /path/to/raw/data` , and change the path of data(pack).
At the beginning of training, I got the msg
import libmesh failed
import libkdtree failed
import utils failed
[image]
I still can continue to train.
But I got an error msg after finishing the first epoch
[image]

I don't know what happened

Steve-Tod, 2021-12-11:
Hi, the import errors are just warnings and that's OK. For the second error, I think the reason is that you didn't generate the occupancy data or the occupancy data is broken. You can follow [here](https://github.com/UT-Austin-RPL/GIGA#save-occupancy-data) to regenerate the occupancy data.

liuallen871219, 2021-12-12:
Thanks for your reply. that's very helpful
I have followed the instruction to generate the occupancy data.
I changed the value of num_worker (data loader) to 1.  The error is gone.
I think it might be the multiprocessing issue for the dataloder.

Steve-Tod, 2021-12-12:
Not sure why that happens. But good to know it works at the end😊.

closed 2021-12-12 by Steve-Tod