UT-Austin-RPL/GIGA #11: visualize grasp afforance

ccc1711, 2022-03-12:
May I ask what software did you use to visualize the grasp afforance? 
colored like this。。。
[image]

Steve-Tod, 2022-03-12:
Hi, you can refer to this code snippets: https://github.com/UT-Austin-RPL/GIGA/blob/d67c4388d334babe6c11c1555a6e848fb4828c84/src/vgn/utils/visual.py#L101

ccc1711, 2022-03-15:
ok，and how can I get the size of feature map for each layer?

Steve-Tod, 2022-03-16:
I think you can print the size of the different layers of the network and compute the size of the feature map.

ccc1711, 2022-03-20:
Is the “TSDF-3DCNN-PointNet-3DUnet” process reasonable?

Steve-Tod, 2022-03-30:
Sorry for the late reply, I didn't see this earlier. What do you mean by “TSDF-3DCNN-PointNet-3DUnet”?

ccc1711, 2022-03-31:
It's nothing ,i was wrong, and  is the process of projection and aggregation implemented in  **def generate_frid_features()**?

Steve-Tod, 2022-03-31:
Yeah, that's right.

closed 2024-02-28 by Steve-Tod