UT-Austin-RPL/GIGA #15: New dataset

ishikaa25, 2022-06-15:
Hi,

I would like to use the pre-trained GIGA model on a custom dataset. 
How have you processed the TSDFs (specifically, `pos_occ`)? 
(I have tried to use RGB-D images -> ScalableTSDFs using Open3D)

Steve-Tod, 2022-06-15:
Hi, 
1. The `inputs` should be the TSDF voxels after fusion.
2. The `pos` is the coordinate of the point where we want to query the grasp parameters.
3. Occupancy points are randomly sampled inside the volume. The occupancy values are queried from the ground truth mesh, like [here](https://github.com/UT-Austin-RPL/GIGA/blob/d67c4388d334babe6c11c1555a6e848fb4828c84/src/vgn/utils/implicit.py#L92). I'm not sure if you really need the ground truth occupancy values if you only want to run the pre-trained GIGA.

ishikaa25, 2022-06-21:
Hey,

Given I have the TSDF voxels in (40,40,40) format, I would like to know how to:
1. Sample points or generate `pos` for querying grasp params? According to your dataset and code it seems like there was a `grasps.csv` from where `x, y, z` were taken and then scaled to match to voxel size. How did you generate the `x, y, z` in the csv file?
2. For running the pre-trained model with my own script, what is the input: `p_tsdf`? (assuming I now have both TSDF voxels and the query point)

ishikaa25, 2022-06-22:
**UPDATE**
I was able to sample random points and generate grasps on my data.

Now, I would like to know:
1. The sizes/shapes of `qual_vol`, `rot_vol`, `width_vol` are `(num_query_points,1)` **each** right?
RPL/GIGA/blob/d67c4388d334babe6c11c1555a6e848fb4828c84/src/vgn/detection_implicit.py#L50
Because in the next line they are reshaped into `(40,40,40)`?

ishikaa25, 2022-06-22:
**UPDATE**

I was able to generate grasps using the above^.
Right now I would like to know how I can visualize the grasps. From the code, I see that you generate a scene mesh. Is there way to visualize grasp on pointcloud data or TSDF voxels?

Steve-Tod, 2022-06-24:
Hi, for grasp data generation, please refer to [this file](https://github.com/UT-Austin-RPL/GIGA/blob/main/scripts/generate_data_parallel.py). We follow VGN and randomly sample points on the point cloud observation.
https://github.com/UT-Austin-RPL/GIGA/blob/d67c4388d334babe6c11c1555a6e848fb4828c84/scripts/generate_data_parallel.py#L76

closed 2022-06-30 by ishikaa25