UT-Austin-RPL/GIGA #16: Visualizing on custom dataset

ishikaa25, 2022-06-23:
What is the coordinate system / reference you take for adding the grasp visual in `grasp2mesh()`? 

I tweaked your code a bit to run it on my dataset, but the gripper is way farther from the object mesh.

Steve-Tod, 2022-06-24:
I think it should be in [-0.5, 0.5] normalized coordinates.

ishikaa25, 2022-06-24:
> I think it should be in [0, 0.3] * 3 coordinates.

You mean the scale of range of the grasp query or the scene?

Steve-Tod, 2022-06-24:
> > I think it should be in [0, 0.3] * 3 coordinates.
> 
> 
> 
> You mean the scale of range of the grasp query or the scene?

The scale of the scene.

ishikaa25, 2022-06-24:
> > > I think it should be in [0, 0.3] * 3 coordinates.
> > 
> > 
> > You mean the scale of range of the grasp query or the scene?
> 
> The scale of the scene.

Right.
So basically I made the TSDF (voxels) such that each voxel has a size 0.0075 cm and so the TSDF has length of 30cm in a (40,40,40) grid.

But in Trimesh the origin is shifted (of the scene), so how did you scale it in your case? Is this just a trimesh thing?

Steve-Tod, 2022-06-24:
> > > > I think it should be in [0, 0.3] * 3 coordinates.
> 
> > > 
> 
> > > 
> 
> > > You mean the scale of range of the grasp query or the scene?
> 
> > 
> 
> > The scale of the scene.
> 
> 
> 
> Right.
> 
> So basically I made the TSDF (voxels) such that each voxel has a size 0.0075 cm and so the TSDF has length of 30cm in a (40,40,40) grid.
> 
> 
> 
> But in Trimesh my scene seems to be shifted from the origin, so how did you scale it in your case? Is this just a trimesh thing?

Does it look like a uniform shift?  Because the prediction space of implicit functions is [-0.5, 0.5], I scale and shift the output into [0, 0.3] in the visualization function.

ishikaa25, 2022-06-27:
Thanks for the help! To sum up the ranges:
> input query points : `[0,0.3]`
> output of implicit function : `[-0.5,0.5]` which is then scaled & shifted to get it in `[0,0.3]` 

Is this correct?

Steve-Tod, 2022-06-27:
> Thanks for the help! To sum up the ranges:
> 
> > input query points : `[0,0.3]`
> > output of implicit function : `[-0.5,0.5]` which is then scaled & shifted to get it in `[0,0.3]`
> 
> Is this correct?

Yes!

ishikaa25, 2022-06-27:
So right now,
My grasp query points/centers are generated like this: 
`pos = np.random.uniform(-0.5, 0.5, (self.num_point,3))`

My actual (input) data is as follows:
1. Depth image: `1280x720 px`
2. TSDF: 
 a. `resolution = 100` --> `(40,40,40)` [truncated manually using numpy slicing since the objects always only lie in this range anyway]
 b. `length = 0.75`
 (Thus, I have a `(40,40,40)` TSDFs voxels, each of length `0.0075` (same as yours)
3. Pointcloud: points scaled from `[0, 0.75]` --> `[0, 0.3]`
4. Meshes: `extents = [0.745576, 0.304891, 0.483592]` 
(So, maybe because I picked the resolution as 100, the mesh extends upto 0.0075*100 = 0.75

How I visualize (code in first comment):
1. I use the GIGA-pile pretrained model to save best grasps (`quality`, `rot`, `width` and `pos[best_quality]`) from the input TSDF voxels and `pos`
2. The grasp centers I get are in the range `[0, 0.3]`, hence I skip the step of scaling and transformation.
3. The remaining values I've used are the same as in your code yet scaling and orientation of the gripper wrt scene mesh are totally off (not uniform for all scenes). 

It'd be great if you could help me out on this an give your suggestions!

Steve-Tod, 2022-07-05:
Hi, it's OK to sample points inside the object, because in that case, the gripper would collide with the object before getting to the target position, and the current data generation script would detect that collision and mark the grasp as a failure.

closed 2022-08-09 by ishikaa25