UT-Austin-RPL/GIGA #20: how to query at a higher resolution of 60×60×60

jxtxwsbn, 2022-08-29:
Hi @Steve-Tod , I have two questions:

1. how to query at a higher resolution of 60x60x60? (I have tried directly set the resolution as 60 in the `detection_implicit.py`, but ```qual_vol[valid_voxels == False] = 0.0``` gives me an error that *the size of valid voxels doesn't match the size of qual_vol*)
2. Without *sideview*, the performances decrease a lot. Do you have any explanations?

Thanks!

Steve-Tod, 2022-08-29:
1. It's probably because you either forgot to specify the resolution of TSDF or the voxel query grid. You need to set both. To set TSDF resolution, see [here](https://github.com/UT-Austin-RPL/GIGA/blob/d67c4388d334babe6c11c1555a6e848fb4828c84/scripts/sim_grasp_multiple.py#L38) and [here](https://github.com/UT-Austin-RPL/GIGA/blob/d67c4388d334babe6c11c1555a6e848fb4828c84/src/vgn/experiments/clutter_removal.py#L37).
2. If you only remove the `--sideview` argument, the model will get one random view observation of the scene, while the model is trained on a specific sideview, so it's natural that it can't generalize. If you don't want `--sideview`, you need to train one model without it to align training and test settings.

jxtxwsbn, 2022-08-30:
Thanks, it works well now. I also noticed that the model fails to generate any grasps especially on the pile scenario. Does it mean the query points are all far away from the observed surface?

Steve-Tod, 2022-08-30:
Hmm I don't think so, the query points are designed to be near the surface from [this line](https://github.com/UT-Austin-RPL/GIGA/blob/d67c4388d334babe6c11c1555a6e848fb4828c84/src/vgn/detection_implicit.py#L135). It's probably the network has low confidence on the predicted grasp (lower than the threshold), you can try using a smaller threshold by changing [this argument](https://github.com/UT-Austin-RPL/GIGA/blob/d67c4388d334babe6c11c1555a6e848fb4828c84/scripts/sim_grasp_multiple.py#L92).

jxtxwsbn, 2022-08-30:
I see. Since I used the `--force` argument, the `qual-th` doesn't affect but there is also a LOW_TH that does affect. Thanks.

closed 2022-08-30 by jxtxwsbn