UT-Austin-RPL/GIGA #21: Installation error:  "  LINK : fatal error LNK1181: can not open input file“m.lib”  "

zqnss666, 2022-11-05:
Hello! I just began my robotics research which is very related to yours' . I am trying to follow your work in Github.
When I carried out the step 4 of Installation, firstly it said the cl.exe could not be found, so I installed Visual Studio 2017 and set the path of environment variables, like following:
env variable **"path"**:
[image]
env variable **"LIB"**:
[image]
env variable **"INCLUDE"**:
[image]

But an error still occurred: **LINK : fatal error LNK1181: can not open input file“m.lib”** . The specific output in the command line:
[image]

I thought it might still caused by libpath or something, but I could not fix it after trials. I think it will definitely affect the following steps of installation or data generation so it is annoying.
Do you have some ideas or suggestions to fix it? Thanks a lot!

zqnss666, 2022-11-08:
Ok，thanks! I have just fixed the error by adjusting the codes(about libraries). But there will be other problems. I would consider carrying out experiments in Linux. So you recommend to use WSL on win10? I do not know if it will fit well with my NVIDIA.

Steve-Tod, 2022-11-08:
I remember installing nvidia driver and cuda-related libraries on WSL on win10 and it works quite well.

zqnss666, 2022-11-09:
I will try. Thanks!

closed 2022-11-19 by zqnss666