UT-Austin-RPL/GIGA #22: About NVIDIA Driver in WSL2

zqnss666, 2022-11-14:
I have tried to install the program in wsl2 using ubuntu. If I use command "nvidia-smi' in linux, I can get a result:
[image]

But when I try to install cuda from nvidia by 
wget https://developer.download.nvidia.com/compute/cuda/11.5.0/local_installers/cuda_11.5.0_495.29.05_linux.run
sudo sh cuda_11.5.0_495.29.05_linux.run
It will warn like that:
[image]

Thus is it necessary to download and install the **nvidia driver for linux 64-bit** in nvidia website? I tried to do that, but I can not install it locally in wsl2-ubuntu system. It reminded that there is not an NVIDIA GPU in this system. Thanks!

Steve-Tod, 2022-11-14:
At the end of the log you can see that you can use `sudo cuda_11.5.0_495.29.05_linux.run -silent -driver` to install driver. And from `nvidia-smi` it seems you already have cuda driver installed. You can try install pytorch and run `a = torch.randn(3).cuda()` and see if there's any error.

zqnss666, 2022-11-15:
Thus if I run it smoothly, it proves cuda will be used in experiments without problem?

zqnss666, 2022-11-15:
And when I tried to build the ConvOnets as step 4 in installation, some warnings happened:
[image]
If it was caused by differentt versions of related packages? And it will affect future experiments like generating or training? Thanks!!

Steve-Tod, 2022-11-15:
The warning is normal. It won't affect the experiments.

zqnss666, 2022-11-16:
Thanks！I am now tring to process data and train the models.

closed 2022-11-19 by zqnss666