UT-Austin-RPL/GIGA #24: [Question] How to get the 3d reconstruction at inference time?

chisarie, 2023-05-08:
Hi, thank you very much for releasing this project!

I have a quick question regarding inference with GIGA. I see that I can predict grasps by using the `VGNImplicit` class and calling its `__call__` function. When visualize is set to `true`, that function returns `grasps, scores, toc, composed_scene`, see here:
https://github.com/UT-Austin-RPL/GIGA/blob/d67c4388d334babe6c11c1555a6e848fb4828c84/src/vgn/detection_implicit.py#L83

But from my understanding, the returned `composed_scene` is just the input tsdf + the added grasps, it does not contain the 3d reconstruction. Is that correct?

So my question is, how do I get the 3d reconstruction? To me it seems that the `p_tsdf` argument and the `self.decoder_tsdf` are relevant here, but I am not sure how to use them. 
https://github.com/UT-Austin-RPL/GIGA/blob/d67c4388d334babe6c11c1555a6e848fb4828c84/src/vgn/ConvONets/conv_onet/models/__init__.py#L64

Assuming this is the right way to get the 3d reconstruction, what should I pass as `p_tsdf`? And then, how do I process and visualize the output of `decoder_tsdf`? 

Thank you very much in advance, I am looking forward to hearing from you!

chisarie, 2023-05-17:
Thank you for the fast response. What is the argument `pc_in` exactly? Is it the raw pointcloud from the camera, including all objects and the table they are sitting on? 
I am trying to do inference from a depth camera I have in real life, not from the simulation

Steve-Tod, 2023-05-17:
Hi, we crop the raw point cloud to remove the table surface and only keep the objects.
https://github.com/UT-Austin-RPL/GIGA/blob/d67c4388d334babe6c11c1555a6e848fb4828c84/scripts/construct_dataset_parallel.py#L34

Steve-Tod, 2023-05-20:
Yes, I think it's correct. I just re-ran the code and found input to `generate_mesh` is with size `torch.Size([1, 40, 40, 40])`, so it's a tsdf grid.

closed 2023-05-23 by chisarie