UT-Austin-RPL/GIGA #25: No module named vgn

HongqingThomas, 2023-09-09:
when I try to run the command:

python scripts/sim_grasp_multiple.py --num-view 1 --object-set (packed/test | pile/test) --scene （packed ｜ pile) --num-rounds 100 --sideview --add-noise dex --force --best --model /path/to/model --type (vgn | giga | giga_aff) --result-path /path/to/result

it shows an error that no module named 'vgn', it seems like sim_grasp_multiple.py fails to import src/vgn or any file under vgn, I'm wondering how to fix it.

Steve-Tod, 2023-09-09:
Hi, did you follow the [installation instructions](https://github.com/UT-Austin-RPL/GIGA#installation)? After installing the required packages, you need to run `pip install -e .` to install the library as a local package.

HongqingThomas, 2023-09-10:
Thanks for the suggestion, now the problem is solved.

closed 2023-09-10 by HongqingThomas