UT-Austin-RPL/GIGA #28: Train GIGA

buaawzh, 2023-09-13:
When I trained the GIGA, I have this problem

  File "/home/wzh/GIGA/src/vgn/dataset_voxel.py", line 98, in read_occ
    path_idx = torch.randint(high=len(occ_paths), size=(1,), dtype=int).item()
RuntimeError: random_ expects 'from' to be less than 'to', but got from=0 >= to=0

I don't know what cause this error.
Maybe I failed to save the valid occupancy data?

Steve-Tod, 2023-09-13:
I think it means the length of `occ_paths` is 0. And `occ_paths` is from [here](https://github.com/UT-Austin-RPL/GIGA/blob/d67c4388d334babe6c11c1555a6e848fb4828c84/src/vgn/dataset_voxel.py#L96). Maybe you should check the directory to see if the occupancy data is saved.

closed 2024-09-06 by Steve-Tod

JohnBrann, 2025-08-14:
I got the same issue, however the data I am trying to train from is directly from the github wget download link. I downloaded the folders with the data inside, ran this command:
`python scripts/train_giga.py --dataset /GIGA/data/pregenerated_data/data_packed_train_processed_dex_noise/ --dataset_raw /GIGA/data/pregenerated_data/data_packed_train_raw/`

This is the what the data looks like:
`/GIGA/data/pregenerated_data/data_packed_train_processed_dex_noise# ls
grasps.csv  point_clouds  scenes`

Do I need to do something else?