UT-Austin-RPL/GIGA #31: The program that generates data gets stuck in the first loop

Jinbo-Zuo, 2023-09-20:
When I run the program that generate-data as instructed in the Readme.md, it got stuck on the first loop. Then I added print to locate where the program was stuck, and found it at the following location:

generate_data.py -- "tsdf = create_tsdf(..."
perception.py -- "tsdf.integrate(depth_img[i], instrinsic, extrinsic)"

Furthermore, in the integrate function,

"self._volume.integrate(rgbd, intrinsic, extrinsic)"

I don't know why I'm stuck here, has anyone else encountered this problem?

Replenish:
1. My running environment is Ubuntu18.04+python3.8
2. This section is very similar to vgn, but vgn is not stuck, even if their input parameters at 'create_tsdf' are the same.

AlchemicRonin, 2023-10-15:
Same problem, but still no idea how to fix it.

eric895888, 2024-04-25:
I have the same problem. I tried adding multiprocessing.set_start_method('spawn') at the beginning to specify the spawn mode to establish multiprocessing. Although it is slower, it works for me. I think self._volume.integrate(rgbd)  , intrinsic, extrinsic) in open3d has an unknown interference with multiprocessing

closed 2024-09-06 by Steve-Tod