UT-Austin-RPL/GIGA #35: Did you consider trying to avoid using the grasp data on the wrong voxels?

Fanqyu, 2024-03-11:
Hi, thanks for your great work!
I wonder if the training dataset includes those grasp data pairing with those ambiguous TSDF voxels, since you fuse the TSDF only by one single-view depth image, which definitely includes the ambiguous voxels in the occluded area. Intuitively, when you take as input those voxels with corresponding grasp data on them, this seems to affect the performance because of the ambiguous dataset.
Have you ever considered this, or already utilized a few tricks in this work to avoid this? It's important to me.
Thanks a lot.

Steve-Tod, 2024-03-11:
Hi, we intentionally train our model with grasp data on the voxels that are invisible, because we wanted to explore the synergy between 3D reconstruction and grasp affordance. Our key intuition is that a learned representation capable of reconstructing the 3D scene encodes relevant geometry information for predicting grasp points and vice versa.

In short, we didn't avoid using the grasp data on the ambiguous voxels.

Fanqyu, 2024-03-12:
Thank you for answering my question. It's very helpful!

closed 2024-03-12 by Fanqyu